harvie256/midi-commander-custom #1: Ability to revert back to original firmware?

snhirsch, 2021-07-09:
Once your new firmware is flashed, is it possible to revert back to factory firmware via DFU?  I'm guessing yes, but would like to know for sure whether bridges get burned. 

Given the amount of aftermarket firmware available for the Behringer FCB1010, it's great to see similar doors opening for the Melo Audio unit.  I'm in the midst of a heavy playing schedule, but would like to dip my two into the waters if safe to do so without bricking the devices.

harvie256, 2021-07-10:
Short answer:
Yes, and I have done it many times to my own device.  However I have no idea if there's any variations on the hardware out there at this point.  So if you're relying on it for your current playing, I wouldn't at this stage.  More testing is needed.

Longer answer:
There's two memories of interest, the flash that's holding the firmware and the EEPROM that holds the configuration. For the standard melo firmware both need to be correct for it to boot.

The flash is programed with a derivative of the STM32 DFU firmware. Unless something is loaded that intentionally over-writes the bootloader area of flash (which there are locks that need to be overridden to do, so it would have to be intentional),  then you'll always be able to reload the flash.

The EEPROM is a little bit more susceptible to problems.  My investigations have shown the current melo firmwares only use the first 128 bytes of the EEPROM, so I'm using the free space above that.  I've also include the capability to backup your EEPROM and restore the original.  So in theory (and tested on my own device), you can swap between the firmware without losing settings of either... 

But there's still some risk there as it hasn't been tested widely yet.

snhirsch, 2021-07-10:
I have a lull in the action after July and will definitely make time to play with this at that point.  In particular, I find the ability to send only 0 or 127 to a CC# to be highly limiting.

harvie256, 2021-07-10:
Great, will be excellent to get some feedback.  Hopefully by then much more testing will have been carried out.

ffabreti, 2022-05-04:
Hi,  I am not sure how to backup and restore the firmware thru `dfu-util` could you give the exactly cmd?
In `man dfu-util` there are some examples, but not sure how it applies to this project:
```man
   Using dfu-util with a DfuSe device
       Flashing a .dfu (special DfuSe format) file to the device:
         $ dfu-util -a 0 -D /path/to/dfuse-image.dfu

       Reading out 1 KB of flash starting at address 0x8000000:
         $ dfu-util -a 0 -s 0x08000000:1024 -U newfile.bin

       Flashing a binary file to address 0x8004000 of device memory and ask the device to leave DFU mode:
         $ dfu-util -a 0 -s 0x08004000:leave -D /path/to/image.bin
```

ffabreti, 2022-05-04:
> The EEPROM is a little bit more susceptible to problems. My investigations have shown the current melo firmwares only use the first 128 bytes of the EEPROM, **so I'm using the free space above that**. I've also include the capability to backup your EEPROM and restore the original. 

Can you please clarify if this project is using the EEPROM ?  You told me in another issue that it is not.
And also, what would be the steps to back it up  using `dfu-util` ?

harvie256, 2022-05-04:
An early version of this project (you'll note that comment was from almost a year ago), used the EEPROM to store the settings in the way described.  This was dumped in favor of the flash in order to store the much larger number of commands as I described in that other thread.

There is no code in either this firmware, or the DFU bootloader that can interact with the EEPROM in any way.  I do see there is an out of date comment in the features list stating that it has SYSEX messages to backup and restore the EEPROM. These were all removed when the code was removed that interacted with the EEPROM.

harvie256, 2022-05-04:
> Hi, I am not sure how to backup and restore the firmware thru `dfu-util` could you give the exactly cmd?
Download the firmware:
dfu-util -a 0 -D generated-20220424-163714.dfu

I've not used it to create a backup, but I see no reason why you couldn't.  You can download all the MeloAudio firmwares from their website, so I never needed to.

closed 2022-05-05 by harvie256