harvie256/midi-commander-custom #2: Ability to reset CC switches on PC issued

redcloud80, 2021-08-10:
Hi! Thanks for this great contribution! I contacted MeloAudio in order to ask for a new custom firmware but they never gave me significant feedback. I can give a hand in both coding and testing.
The changes I asked are well explained in this video

https://www.youtube.com/watch?v=tF4gexntm0s

I need a firmware that reset the status of upper switches (1, 2, 3, 4) when a lower switch (A, B, C, D) is pressed. I use A, B, C, D to switch presets and 1, 2, 3, 4 to send CC.

As an example, I use switch A for a clean preset and switch B for a crunch preset. When switch B is on, if I press switch 1 it activates a boost. So please follow this steps:

- Current status is (switch A led=on, switch B led=off, switch 1 led=off, CC#22 value=0)
- Press switch B (switch A led=off, switch B led=on ,switch 1 led=off, CC#22 value=0)
- Press switch 1 (switch A led=off, switch B led=on ,switch 1 led=on, CC#22 value=127)
- Press switch A (switch A led=on, switch B led=off,switch 1 led=off, CC#22 value=?)
- Press switch B (switch A led=off, switch B led=on,switch 1 led=off, CC#22 value=?)
- Press switch 1 (switch A led=off, switch B led=on ,switch 1 led=off, CC#22 value=0) <----- wrong! I expect (switch A led=off, switch B led=on ,switch 1 led=on, CC#22 value=127)

harvie256, 2021-08-10:
Thanks, you're more than welcome to contribute wherever you'd like.

Currently I'm moving interstate and switching jobs so it's unlikely I'll get back to working on this for a couple of months.  

My next real challenge for it was to build a desktop app to aid configuration, as with all the new requirements I've gathered from helpful folk it's badly out grown the use of a spreadsheet for configuration.  This would also involve switching to variable length command sets.

So unfortunately I can't help you much for a little while, as I don't have the spare time.

redcloud80, 2021-11-21:
I can help you more on the desktop app side since STM32 is unknown to me.

davidin73, 2022-06-12:
Quote.....😅

Charles5150, 2026-09-26:
In case anyone still needs this, it can be done now in the fork I maintain (built on arasan95's, which builds on this one).

Two pieces cover the steps you describe:

- **Exclusive groups**: put A, B, C and D in one group and switching one on switches the others off, LEDs included. [Docs](https://github.com/Charles5150/midi-commander-custom/blob/main/docs/manual/en/05-buttons.md#exclusive-groups)
- **Scenes**: a single command that puts the bank's toggles into a chosen state. Add one to A and B that switches 1 off, and the boost LED goes back to off together with the preset change, so the next press of 1 sends 127 again. [Docs](https://github.com/Charles5150/midi-commander-custom/blob/main/docs/manual/en/05-buttons.md#scenes)

If your device reports its state over MIDI, `LED_Feedback` can make the LEDs follow it instead.

Ready built image: https://github.com/Charles5150/midi-commander-custom/releases/latest