harvie256/midi-commander-custom #5: Uploading the firmware

thesoulharmonic, 2021-08-21:
Hi there,

Firstly, thank you for programming this as it has given me some hope that I may be able to configure the MeloAudio to do what I need (Control and SP-404).

I'm not really a big coder but I think I've managed to create some control in the spread sheet and get them converted with the python script.

I was wondering if you'd be able to help me with the actual uploading of the firmware to the device as I can't seem to find a .dfu file as described in their manual.

Am I right in thinking you can edit the spreadsheet with the notes and commands you want then pass that through the python script and then upload that to the device and it should work?

Any help would be much appreciate.

thesoulharmonic, 2021-08-21:
This is the section I'm struggling with 

#Other than the simple python scripts, it's all just STM32CubIDE. Install that, import the project into your workspace (it's just shrink wrapped eclipse) and your done.

There's two Build settings, one for the DFU (with offset linker script and vector table) and the other for using with a ST-Link debugger. To build a DFU file for upload you'll need to build the binary in the IDE, then use the DFU packing tool that comes with the DFU uploader (can't remember their exact names off the top of my head.) Using the ELF file instead of the .bin saves you having to input the flash offset.

thesoulharmonic, 2021-08-23:
Just to give you an update I managed to create a .dfu file using the 'DfuFileMgr' from the file 'MIDI_Commander_Custom.bin' file and then load this via the **DfuSe Demo** but it it hasn't loaded the custom firmware to the device. I loaded the custom dfu file to the 00 Internal Flash

Am I missing something?

harvie256, 2021-08-23:
Sorry I didn't see your issue earlier, I'm not working on this at the moment (I wont even have access to the hardware for a few months).

This is not in a release state yet, and as such is really only suitable for those wishing to continue development.  That is the intention of those "Build" instructions.  I apologize for not including this in the Readme, I will update it.

When I got close and asked for feedback, I received a large number of feature requests so figured it was not worth trying to finalize a tested build when (1) There wasn't much interest in it with the features it had, only future prospects (2) no one was interested in contributing.

I could build and upload a DFU from the current code base, but I really have no idea how well it would or wouldn't work as I don't have access to any hardware to test.

thesoulharmonic, 2021-08-24:
Ah I see. Well yeah if you wouldn't mind uploading a test release dfu I'd still like to give it a go and test it. Thank you

harvie256, 2021-08-24:
I've built the current code and added a new folder to the repo
test_dfu_image/test_image_2021_08_24.dfu

In theory you should be able to upload this, and on boot get a splash screen with the firmware build "0.1A".
Then use the CSV_to_Flash.py to upload the configuration into flash.

Again I have no ability to test any of this at the moment, so I'm not that confident it'll work for you.

Best of luck!

thesoulharmonic, 2021-08-24:
Great! I will report back. Thank you

redcloud80, 2021-11-09:
I'm trying to flash test_dfu_image/test_image_2021_08_24.dfu but when I turn on the device, nothing happens and all LEDs remain turned on. No splash screen at all.

alefnula, 2021-12-05:
Hi @redcloud80 How did you manage to flash the MP100? Holding down the switches while turning it on does not put mine in the update mode.

redcloud80, 2021-12-06:
If I correctly recall you must keep pressed both D + BANK DOWN buttons then power on.

alefnula, 2021-12-06:
That's what I tried but I just get a blank screen and no midi device is registered on the computer :-/

davidin73, 2022-03-06:
> That's what I tried but I just get a blank screen and no midi device is registered on the computer :-/

Hello!!! Were you able to install the firmware in the end?

alefnula, 2022-03-09:
@davidin73 No, I just gave up in the end. I don't think it works with the HarleyBenton version.

eliericha, 2022-03-12:
@alefnula You may have encountered the same issue I did in #9 . The firmware refuses to start if your board has less than 512kB of flash. Mine had 256kB. But in practice I found that 256kB were enough to house the firmware and data so I changed the code to allow 256kB as a minimum. My version is in pull request #14 . It also increases the number of commands that can be configured on each button press to 10.

But I didn't provide a DFU file ready to upload because I'd rather get my code vetted by @harvie256 first :) Besides, I'm on macOS and I couldn't find any tools to create a DFU file.

eliericha, 2022-03-12:
@davidin73 I know you commented on my pull request, but I'll answer here since this issue is a better place for that topic :)

Are you on Windows, Linux or macOS ?

If you're on Windows, you need to use the DFU utilities that are in this repository under the DFU directory. The MeloAudio firmwares contain some basic instructions on how to do it. So I suggest you download one from [their website](https://meloaudio.com/pages/manual) and try with their firmware. If that works, then you can come back here and try loading the DFU file that @harvie256 provided.

davidin73, 2022-03-12:
> @davidin73 So che hai commentato la mia richiesta pull, ma risponderò qui poiché questo problema è un posto migliore per quell'argomento :)
> 
> Sei su Windows, Linux o macOS?
> 
> Se sei su Windows, devi usare le utilità DFU che si trovano in questo repository nella directory DFU. I firmware MeloAudio contengono alcune istruzioni di base su come farlo. Quindi ti suggerisco di scaricarne uno dal [loro sito Web](https://meloaudio.com/pages/manual) e provare con il loro firmware. Se funziona, puoi tornare qui e provare a caricare il file DFU fornito da @harvie256 .

Hi!!! I'm on win 10,and i have a Harley Benton controller.
Okay, i must find the time and if it works i try with the Harvey's file and post the result.

closed 2022-05-05 by harvie256