harvie256/midi-commander-custom #9: Board freezes with all LEDs on at startup

eliericha, 2022-02-17:
Hello,

First off thank you for developing this alternate firmware :) I bought my MIDI Commander a while ago and I was quickly surprised by the limited configurability of the original firmware. I was excited to discover your project last week and I gave it a try!

I managed to build the binaries in DFU Release mode in STM32CubeIDE.

After that I struggled to load the firmware on the board because I'm on macOS where the DFU management Windows executables are unusable. It took me a while before I realized that I could load the .bin file if I specify the load address to be 0x8003000. Can you confirm that this is correct?

I then found another way to produce the DFU file. The [dfu-util](https://github.com/kiibohd/dfu-util) project has a Python script that allowed me to create the DFU file from the .hex file.

However when I loaded the firmware (with either methods) the board started with all LEDs turned on and nothing happened. So surely there's something happening early on.

Any ideas how I could investigate what's going on?

I found the main() entry point, but I don't know what I could insert there to see if the board is at least able to run the entry point.

Now I'm also wondering if perhaps I have a different version of the product than yours? Is it possible that they've changed something in the hardware that made it incompatible with the configuration of your project?

If you still have some motivation for this, I would love to collaborate somehow :) I'd love to finally make the pedal usable in my setup. My end goal is to control the effects section of my new Boss RC-600 loop pedal by sending multiple MIDI messages per pedal.

Thanks!

eliericha, 2022-02-17:
I already have news.

I figured out how to control LEDs, so I wrote a line of code to turn off one of the LEDs, and by moving that statement along the main control flow I found that the problem is the check for flash size.

```
  uint16_t flash_size = (*(uint16_t*)FLASHSIZE_BASE);
  if(flash_size < 512){
	  Error_Handler();
  }
```

Again by controlling the LED, I found that the flash size of my device is 256k. So indeed less than 512k.

From the main README I see that the config is stored in flash. The `config_image.bin` available in the repo is 896 bytes, so I figured it's safe to proceed and I changed the condition.

I got the display message briefly, then black because there's no config initially. So I used `Dev_Write_Settings_To_Flash.py` to load the the existing `config_image.bin` file and everything worked!

So now I should be able to create my own configuration and load it.

Awesome! Many thanks!

harvie256, 2022-02-20:
Thanks for the feedback, very interesting about the different flash sizes.  Knowing that will change what can be done in future with it.

I intend on coming back to this project and incorporating a lot of new features, however I've been working remotely for around 6 months and haven't had access to the hardware.  Next month I may get the chance to get back to it.

closed 2022-02-20 by harvie256