harvie256/midi-commander-custom #12: More than 3 MIDI commands per switch?

eliericha, 2022-02-27:
Hello there,

As I experiment with controlling the RC-600 I'm finding myself wanting to send more than 3 MIDI messages per switch :)

# Option 1

One way is to keep using the same CSV format and extend it to more commands. Let's do some math to see how much fits in the flash memory. Currently each MIDI command takes up 2 bytes. We have 8 pages of 8 switch of 3 commands each, so:

```
8 * 8 * 3 * 2 = 384 bytes
```

We're still far from the 256 kBytes available on my board.

Ideally I would like to send about 10 MIDI messages per switch (to configure various effects and mute some inputs on the RC-600). So that's:

```
8 * 8 * 10 * 2 = 1280 bytes = 1.25 kB
```

So that should still be ok, right?

# Option 2

Another approach would be to change the format of the data in memory to allow a variable number of commands per switch.

# Option 2.a

One option is to encode commands in memory like strings. For each switch (of each page) we store a sequence of 2-byte commands terminated by a special terminator command (all zeroes?).

Finding switches and pages in memory would mean seeking through the data and counting the special terminators which may be costly to do at each button press (although I'm sure the processor is fast enough for this to be feasible).

But there is possible optimization. We can store a global pointer pointing to the beginning of the active page so that we don't have to seek from the beginning of the config at every switch press.

Similarly, upon page changes, we can update a set of global pointers, one per switch, pointing for each switch to the beginning of the chain of commands.

So page switches would be costly and would involve seeking through the data, but once a page is active, there is no seeking and the data is readily accessible.

# Option 2.b

There is another possibility of encoding the number of commands in the first byte of every switch's chain of commands. The number of commands would be used by the code to jump from switch to switch, and from page to page. It would be slightly more efficient than seeking through the data. The same optimization of maintaining direct pointers to switch commands for the active page, and only doing the pointer jumping upon page changes.

Let me know what you think of these options. From your experience which would be the best bet for me to try?

Thanks!

eliericha, 2022-02-27:
Oops. I miscalculated in option 1. Each command is 4 bytes, not 2.

So for 10 commands per switch it takes:

```
8 * 8 * 10 * 4 = 2560 bytes = 2.5 kB
```

So now the data is beyond one flash page size (1kb) and I think the load code needs to account for that because it seems to assume that the data holds within 1 flash page.

So yeah, I'm experimenting with option 1 since it's the least disruptive to the existing code :)

harvie256, 2022-05-05:
Closed as it's been incorporated.

closed 2022-05-05 by harvie256