harvie256/midi-commander-custom #15: Help understanding Flash Memory of Midi Commander

ffabreti, 2022-05-01:
I have installed DfuSe (3.0.6) from [MeloAudio Manual Page](https://meloaudio.com/pages/manual) and successfully updated to MOME firmware.   Because MeloAudio firmwares only solves one problem at a time (but not all together) it is really limiting. 

Could you help us understanding the 3 Flash memories that show up in DfuSe?  

| Flash Type  |  Sectors | Size/Sector | Total Kb |
|-------------| ----------|-------------|-----------|
| Internal Flash  | 256 | 2kb | 512kb |
| SPI Flash | 128 | 64kb | 8192 kb |
| Nor Flash | 256 | 64kb | 16384 kb |

Where the original firmware is written ?
Where the DFU generated by this project is written ?

ffabreti, 2022-05-01:
Here is a image:

[image]

ffabreti, 2022-05-01:
This is double-clicking Internal Flash:
[image]

harvie256, 2022-05-02:
So I suspect the reason 3 memory types show up is because MeloAudio used the DFU bootloader directly from a STMicro demonstration board and didn't bother making their own. This is also why it shows up in Device Manager as 'TSMIDI' or something similar.

So to answer your question, only the Internal Flash actually exists, the others simply aren't there.  All firmware is written to the internal flash.

In the case of the MeloAudio firmware, they are writing the settings to an external small eeprom memory.  For this custom firmware, we are writing the settings to an unused area in the Internal Flash because there's not enough room in the external eeprom to hold all the button settings for our more complex firmware.

closed 2022-05-02 by harvie256

ffabreti, 2022-05-02:
Thanks. So this project doesn't use de eeprom ?  I mean, the converted spreadsheet is written in the Internal Flash also?

harvie256, 2022-05-02:
That is correct, the eeprom is untouched by this firmware.  This means you can try this firmware and revert back to the MeloAudio firmware without losing the MeloAudio firmware.