harvie256/midi-commander-custom #16: FIX for DfuSe install on Windows 10

ffabreti, 2022-05-01:
This is kind of a helping guide. Those instructions need checking, because I have made then once and did not saved the steps.

[Reference](https://meloaudio.com/community/champ/forums/1764-frequent-questions-for-midi-commander/topics/22311-dfu-mode-not-working-on-windows-10) 

If you're having trouble with DfuSe recognizing your Midi Commander on Windows 10, here are some tips:

TIP 1:
1. UNPLUG your Midi Commander from USB
2. Go to Device Manager > View > Hidden Devices > Other Devices > Look for tsmididfu. 
3. Uninstall/remove this driver
4. Put your hardware in DFU mode by pressing buttons C and D together and power the device. 
5. This allowed windows to properly install DFU mode for me. 

TIP 2:
1. Windows 10 may show a message that the driver is not properly installed
2. Right-click it indicates update the driver > choose Windows update
3. In Windows update you look for updates and then press the option to see optional updates
4. In optional updates you select STMicroelectronics - Other hardware - STM Device in DFU Mode
5. with that it has detected my hardware.

TIP 3:
1. Install Windows DFU Program
2. Go to C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Bin\Driver\Win10
3. run (double-click) dpinst_amd64.exe
4. now the driver should be correctly installed and dfu mode detected

harvie256, 2022-05-02:
Thanks, I'll add these into the document for future.

closed 2022-05-02 by harvie256

ffabreti, 2022-12-26:
You may download DSUSE from https://www.st.com/en/development-tools/stsw-stm32080.html#get-software