harvie256/midi-commander-custom #17: Error running Python script to convert CSV - no input found

ffabreti, 2022-05-01:
Since @harvie256 has provided a DFU file (which I used to update my hardware), I am trying to run [/midi-commander-custom/python/CSV_to_Flash.py](https://github.com/harvie256/midi-commander-custom/tree/main/python) to get the Spreadsheet inside the firmware.

First thing I notice is that after  updating the firmware with [the file in DFU_OUT directory](https://github.com/harvie256/midi-commander-custom/tree/main/DFU/DFU_OUT)  (now, it's generated-20220424-163714.dfu) I get no MIDI output and the hardware screen is off (but pressing buttons do turn the leds on and off).

Maybe it's all expected.

So, I have done these steps in Linux:  
```bash
#in host machine:
docker run -it --detach --name fer-teste debian:11 bash
docker exec -it fer-teste bash

#inside the container:
apt update --allow-releaseinfo-change
apt install git python3 python3-pip  python3-wheel python3-dev build-essential
apt install libjack-dev libasound2-dev
git clone https://github.com/harvie256/midi-commander-custom.git
cd midi-commander-custom/python
python3 -m pip install -r ./requirements.txt

python3 ./CSV_to_Flash.py ./'MeloConfig_10_Cmds - RC-600.csv'
#error returned:
=> no input found
```
Any advises ?
What am I missing ?

ffabreti, 2022-05-01:
Fact is you have to run scripts **NOT** in DFU mode.
But, running `CSV_to_Flash.py` produced no results, nor errors.

What worked for me was:
```bash
python3 ./Dev_Convert_CSV_To_Config_Binary.py 'MeloConfig_10_Cmds - RC-600.csv'
#Result = no error messages

vim ./Dev_Write_Settings_To_Flash.py 
    change input file name in variable "File_To_Write"

python3 ./Dev_Write_Settings_To_Flash.py 
	Erasing Flash Settings
	Erase Complete
	Writing Flash Chunk:  1 / 168
	Writing Flash Chunk:  2 / 168
	...
	Writing Flash Chunk:  168 / 168
	Finished, reseting device...
Result = 	hardware operating (sending midi)	
			Screen showing "PRES Presets"
			Buttons 1,4 sending CC Momentary
			Button A sending Several commands
```
I have used file `MeloConfig_10_Cmds - RC-600.csv` present in the repo

ffabreti, 2022-05-01:
I am not sure if this is a problem, but `Dev_Convert_CSV_To_Config_Binary.py` says in the end that the file should be 896 bytes long. Mine isn't .

```bash
tail -n2 Dev_Convert_CSV_To_Config_Binary.py
      # File should be 896 bytes long
ls -la
-rw-r--r-- 1 root root  2688 mai  1 17:53  config_image.bin
-rw-r--r-- 1 root root   896 mai  1 17:00  config_image.bin.original
```

harvie256, 2022-05-02:
You are correct in that DFU mode is a bootloader mode for writing firmware to the flash.  Where as the settings are written to flash by the running firmware, so that needs to boot first.  Also, seeing nothing on the screen when there's no configuration loaded is to be expected. No it's not great and is a feature needed to validate the configuration section with a CRC or similar and display an error message.

harvie256, 2022-05-02:
Those development scripts, looking back I see they weren't updated in eliericha's pull request which expanded the range to 10 cmds per button request.  So they now need to be removed.  So CSV_to_Flash.py is the script you want to be using, the question is why you're getting nothing out of it? I've used it many times under Windows, let me see if I can reproduce what you've got in Ubuntu.

ffabreti, 2022-05-02:
Well, not sure if I messed it up by running `SCV_to_Flash.py` before, but after running `Dev_Write_Settings_To_Flash.py` I did get 6 messages (more than 3) at buttom A press, like this:
```
10:43:03:535 Port1 Ch 5 CC       B4 01 00 [001 Modulation,   0]
10:43:03:535 Port1 Ch 5 CC       B4 02 7F [002 Breath, 127]
10:43:03:535 Port1 Ch 5 CC       B4 0B 00 [011 Expression,   0]
10:43:03:535 Port1 Ch 5 CC       B4 0C 00 [012 Effect 1,   0]
10:43:03:536 Port1 Ch 5 CC       B4 0D 00 [013 Effect 2,   0]
10:43:03:536 Port1 Ch 5 CC       B4 0E 00 [014 ---,   0]
10:43:03:650 Port1 Ch 5 CC       B4 01 00 [001 Modulation,   0]
10:43:03:650 Port1 Ch 5 CC       B4 02 7F [002 Breath, 127]
10:43:03:650 Port1 Ch 5 CC       B4 0B 00 [011 Expression,   0]
10:43:03:650 Port1 Ch 5 CC       B4 0C 00 [012 Effect 1,   0]
10:43:03:650 Port1 Ch 5 CC       B4 0D 00 [013 Effect 2,   0]
10:43:03:650 Port1 Ch 5 CC       B4 0E 00 [014 ---,   0]
``` 
actually there are 12 messages, half sent at press and half sent at release

ffabreti, 2022-05-02:
How can I reset everything and redo the steps only for `SCV_to_Flash.py`?

harvie256, 2022-05-02:
That's not overly surprising, the update from eliericha expanded it to 10 cmds per button.  So you should simply be able to run CSV_to_Flash.py and it should over write the entire settings area.  I just followed exactly the steps you showed in your first post on a fresh Ubuntu VM, with that latest generated DFU firmware image and it worked fine.

$ python3 CSV_to_Flash.py 'MeloConfig_10_Cmds - RC-600.csv' 
Flash content is 2688 bytes = 2.625 kB
Continue? (y/N) y
Erasing Flash Settings
Erase Complete
Writing Flash Chunk:  1 / 168
Writing Flash Chunk:  2 / 168
Writing Flash Chunk:  3 / 168
Writing Flash Chunk:  4 / 168
Writing Flash Chunk:  5 / 168
Writing Flash Chunk:  6 / 168
... Cut out
Writing Flash Chunk:  168 / 168
Finshed, reseting device...

ffabreti, 2022-05-02:
Great! I will test it and report back.

I have tried to run this on Windows 10 WSL2 (Linux Subsystem), but had problems with ALSA drivers (because they were not installed on Windows), did you tested thru a VirtualBox VM or something alike ?

harvie256, 2022-05-02:
Yep, I just use VirtualBox. The dfu-util project works fine for loading the firmware through Linux as well (again tested in a Virtualbox VM.) That's actually how I tend to do it, as I find that Dfuse utility a bit of a pain.

gutomancha, 2022-05-03:
Boa tarde pessoal, não sei se posso perguntar isso aqui, se não puder me avise que retiro, sou usuário de um Mídi Commander, uso ele em uma fractal FM3, meu sonho é uma atualização para enviar Numero CC e comandos CC, essa versão atual que estão mexendo já oferece isso? Outra questão, não sou programador, é complicado eu mesmo atualizar? devo levar para um programador? atualmente utilizo o meu no modo SCENE e alterei a configuração do D para sair de tap para cena +1 afim de usar as 8 cenas, porém, preciso estar na 7 (botão C) e ai o D adiciona +1 levando para o 8. obrigado por enquanto!

harvie256, 2022-05-04:
> Boa tarde pessoal, não sei se posso perguntar isso aqui, se não puder me avise que retiro, sou usuário de um Mídi Commander, uso ele em uma fractal FM3, meu sonho é uma atualização para enviar [...]

Yes it's off topic and you should put in a new thread.  Having said that, I can't recommend you use this project yet if you will have difficulty working through problems.  It's a work in progress.

ffabreti, 2022-05-04:
> Great! I will test it and report back.

It worked. Using VirtualBox and Ubuntu Server 22

```bash
root@ubu:/home/fer/midi-commander-custom/python# python3 ./CSV_to_Flash.py ./'MeloConfig_10_Cmds - RC-600.csv'
Flash content is 2688 bytes = 2.625 kB
Continue? (y/N) y
Erasing Flash Settings
Erase Complete
Writing Flash Chunk:  1 / 168
Writing Flash Chunk:  2 / 168
Writing Flash Chunk:  3 / 168
[...]
Writing Flash Chunk:  168 / 168
Finshed, reseting device...
```

closed 2022-05-04 by ffabreti

gutomancha, 2022-05-04:
> > Boa tarde pessoal, não sei se posso perguntar isso aqui, se não puder me avisar que retire, sou usuário de um Mídi Commander, use ele em uma fractal FM3, meu sonho é uma atualização para enviar Numero CC e comandos CC, esta versão atual que está consumindo já oferece isso? Outra questão, não sou programador, é complicado eu mesmo atualizei? devo levar para um programador? utilizo o meu modo SCENE e alterei a configuração do D para sair de tap para cena +1 afim de usar como 8 cenas, porém, preciso estar na 7 (botão C) e ai o D adicionar +1 levando para o 8. obrigado por enquanto!
> 
> Sim, está fora do tópico e você deve criar um novo tópico. Dito isso, não posso recomendar que você use este projeto ainda se tiver dificuldade em resolver problemas. É um trabalho em progresso.

Muito obrigado pela resposta, vou retirar o tópico