harvie256/midi-commander-custom #18: Suggestion: python config

ffabreti, 2022-05-06:
Hi @harvie256 
What about integrate the CSV file into the repo so that we can track changes ?
I've created a specific dir for configs too.

https://github.com/ffabreti/midi-commander-custom/commit/8ffa8a304cfefe27987b6afce60c7b0eb36cb8e3

ffabreti, 2022-05-06:
I think we can define a empty Google Sheet with formatting for better reading and import/export from there.
So that we can simply paste the CSV in there, and Download later as CSV.

This is possible if we define the lines that will be fixed like : 
- Line 35 must be Button_Settings
- Line 39 must be Bank_Number
- etc

ffabreti, 2022-05-06:
Google Sheets conditional formating of colors, vertical text is fixed for line 39
Could no find where to change idiom to english.

[image]

harvie256, 2022-05-07:
Looks great, I like the colouring. I haven't really played with locking cells in google sheets, if it works then great!

I am slowly working on a desktop GUI app for the configuration to make it easier for folk that find the whole CSV->Python thing a bit much. Also it will allow to more interesting features to come to life.  But that could be a little while off depending on how much free time I actually get to work on it.

Do you want to drop whatever changes into a pull request and we'll incorporate them?

ffabreti, 2022-05-07:
Great, will do more testing to confirm the workflow. 
In what framework are you developing the frontend?  Maybe I can help.

harvie256, 2022-05-07:
C# WPF
I know it's not multi platform, but it's easy to work with...

ffabreti, 2022-05-07:
Didn't know it.  Will take a look at it.  [This guy](https://www.youtube.com/watch?v=gSfMNjWNoX0) says not to start dragging things into the form because it ends up with fixed alignments that may become a nightmare.

harvie256, 2022-05-07:
That's right, it's very similar in ways to coding for Android etc where you're doing the layout in responsive containers using xml.

ffabreti, 2022-05-10:
Hey @harvie256 

I tested the spreadsheet template and decide to go the "Apps Script" route.  Not sure if it was worth it, but it helps.

Take a look at [this commit](https://github.com/ffabreti/midi-commander-custom/commit/4502af9e312a990272030403850c996922bf4636)  and [this sheet](https://docs.google.com/spreadsheets/d/1Zl5tTLALiXYquzPLrpFXYa89CvCHwyR-GLH32q7Evb0/edit?usp=sharing)

User only needs to open google spreadsheet, paste the CSV and paste the script.

Charles5150, 2026-09-26:
A late follow-up on the desktop app mentioned here: the fork I maintain (built on arasan95's) has one now. It is Python, so it runs on Windows, macOS and Linux. It reads and writes the configuration straight from the pedal over SysEx, so there is no spreadsheet or DFU step for a config change, and CSV import and export are kept for version control, which was the original suggestion.

Configurator docs: https://github.com/Charles5150/midi-commander-custom/blob/main/docs/manual/en/03-the-configurator.md