harvie256/midi-commander-custom #22: Black Screen after firmware update

dsovino, 2022-08-18:
Hi, I uploaded generated-20220424-163714.dfu file under DFU_OUT folder to the Midi Commander. 

While booting up I see a "0.1A" message, and after that screen goes completelly black.

Is this normal behaviour or I am missing something here? 

I need badly to able to send Note messages to control my Synthstrom Deluge.  I downloaded the STM32IDE to fork your repo as a starting point. 

Thanks in advance.

Specs:
MacOS Catalina 10.15.7
MacBook Pro (Retina, 15-inch, Mid 2015)

dsovino, 2022-08-18:
Just to add to the conversation. I can succesfully go back to the "Freeze fixed" _dfu_ file from meloaudio website.

davidin73, 2022-08-19:
Look at this issue,is the same problem

https://github.com/harvie256/midi-commander-custom/issues/20

Airwood007, 2022-08-20:
Hi guys, I have no idea how to use github, tried to find a way to message the creator but didnt figure it out.

How do I download this software and use it with my midi commander? Im just trying to get the pedal to work with loopy pro, I can follow instructions if anybody is willing to guide me through getting started, thanks

dsovino, 2022-08-20:
> Look at this issue,is the same problem
> 
> #20

Confirmed. After running the CSV_to_Flash.csv python script it updated the firmware and got it working now. Thanks for the reply

dsovino, 2022-08-20:
Here again with more feedback. 

After succesfully loading the firmware with python script, i went back to the original firmware, make some changes on a Custom CSV and then tried lo load it with python3. 

Now, strangely i get "no input found".

I played a bit with python and make a few changes, just to get some feedback to this:

`# midi_inputs = [x for x in mido.get_input_names() if 'STM' in x]
# midi_outputs = [x for x in mido.get_output_names() if 'STM' in x]

midi_inputs = [x for x in mido.get_input_names()]
print('midi_inputs:', midi_inputs)
midi_outputs = [x for x in mido.get_output_names()]
print('midi_outputs:', midi_outputs)`

If i boot Midi Commander while holding D + Down I dont get any device listed. If i boot it up normally, i get this:

`midi_inputs: ['TSMIDI2.0', 'TSMIDI2.0']
midi_outputs: ['TSMIDI2.0', 'TSMIDI2.0']`

So I guess I have a pretty good confusion about how i need to boot the Midi Commander in order to upload the firmware with the python script. I'm almost 100% sure that the successful upload that i was able to perform yesterday was while having D + Down pressed while botting, but now it doesn't work since it's not listing any input or output device. In any case, not a single "STM" device is listed. 

If I run my version of the script, it just gets stuck on "Erasing Flash Settings", either using input 0 or 1 in:

`inport = mido.open_input(midi_inputs[0])
outport = mido.open_output(midi_outputs[])`

Once again, thanks for your time and support on this.

davidin73, 2022-08-20:
https://github.com/harvie256/midi-commander-custom/issues/17

I think the solution is to run the script with the meloaudio NOT in DFU mode
Look at this issue i have posted

dsovino, 2022-08-21:
[image]

And indeed it worked. Stoked!

davidin73, 2022-08-22:
👍🏻