harvie256/midi-commander-custom #23: How do I download and use this software with my Midi Commander?

Airwood007, 2022-08-20:
Hi guys,

Im completely new to github (got it for this)

Im trying to get Loopy Pro (iOS looping softare) to recognise my Midi Commander through midi learn, the same way it recognizes my other devices. It sounds like if I download and install this software that would work but I cant even see where to begin, if somebody is willing to guide me through getting this midi controller set up so I can use it with Midi Learn im happy to pay them $20 of any cryptocurrency they like.

davidin73, 2022-08-20:
Hi! 
Strange thing, i have read on the net that the midi Commander is recognized from loopy pro🤔
Have you tried with some other software,just to test it?
Do you have the otg cable?

Airwood007, 2022-08-22:
Hi, I ended up returning this pedal and settled on making my own midi controller. I tried and tried to get the pedal to work with LoopyPro and with AUM. I could get AUM to show me that it was detecting an input but it refused to communicate with Midi Learn. I have never had an issue with midi devices and these two apps so I'm fairly convinced its an issue with the way the Midi Commander is set up.

ffabreti, 2023-04-10:
I use it with Loopy Pro.  My setup is:
`IPAD < Lightning Adaptor < Camera Adapter < Usb < UMC204HD < midi IN < MidiCommander`
It works with this firmware.

I thing MidiCommander usb connection is buggy (it freezes), but MIDI cable works ok.

Charles5150, 2026-09-26:
For anyone who finds this later, about the USB freezes mentioned above: the USB side has been reworked in the fork I maintain (built on arasan95's, which builds on this one).

- Outgoing USB MIDI no longer busy-waits for the host.
- Every incoming message, Active Sensing included, is consumed, so the link cannot stall.
- A full buffer no longer hangs the pedal.

I have not tested it with Loopy Pro specifically. It sends plain CC, PC and notes over USB, so MIDI learn should see it like any other controller.

Ready built image: https://github.com/Charles5150/midi-commander-custom/releases/latest