harvie256/midi-commander-custom #27: Any chance of a newbie guide of how to use the spreadsheet and generate a new DFU ?

andrema2, 2023-02-20:
Hi

The Meloaudio site has been down for weeks now. I'm facing freezes on the MIDI Controller and I also would like to change some configurations. I don't know if the DFU that is available here has the correction to the freeze.

Also, the spreadsheet was not clear to me. So any tips would be great. 

Congrats for this major work

ffabreti, 2023-04-10:
Midi Commander has a buggy USB implementation, I use it plugged via MIDI cable and it works great.

Timebutt, 2023-05-22:
Does the updated firmware in this repository address the buggy USB implementation?

harvie256, 2023-05-22:
This firmware isn't an "update", i.e. It's not like Meloaudio gave me their code base for me to work from. It's created from reverse engineering the hardware and building something new from scratch.

I've only done fairly basic testing of the implemented USB interface, as I don't tend to have a need for it. So I don't have any idea how reliable the midi side of the USB is in practice.

Timebutt, 2023-05-23:
Just to answer my own question: I flashed `TSMIDI_DFU_V3_STD2.dfu` and it made no difference in terms of USB connectivity: it still freezes whenever I open up Cubase.

Timebutt, 2023-05-23:
As a final update: I finally finished flashing the `DFU` from this repository and configuring the CSV on my device and holy moly: IT WORKS! The device no longer freezes when I open Cubase and is entirely stable when running over USB. @harvie256 thank you for your awesome work, you just made sure I don't need an external MIDI cable and interface to hook this up to my Mac for live shows ;)

eliericha, 2023-08-18:
Hello there,

I've been improving the documentation a bit on [my fork](https://github.com/eliericha/midi-commander-custom/tree/main), in particular with some more details about installing Python dependencies and loading the firmware from macOS.

If you had trouble with Python 3.11 then I guess we should recommend Python 3.9 in the documentation.

rahmatismail, 2025-04-26:
Hello everyone. Can i have a copy of the spreadsheet? Seems like the file is deleted

harvie256, 2025-04-26:
@rahmatismail Seems the google sheets link has disappeared. The CSV file is still there but that doesn't have any of the fancy styling.

I haven't worked on this project in a very long time, or used it for that matter. Anyone have a copy we can share?

ivaylomil, 2025-04-27:
Hi!
I have a copy modified for my needs around hx stomp.
It may not have the latest formatting.
[Copy of MeloConfig HX Stomp](https://docs.google.com/spreadsheets/d/164-Dd9iwLZsoiOmuHsXUMjPhh0ImSzeXSqe5HUqpTaM/edit?usp=sharing)
@rahmatismail, let me know if it works for you.
Then it could be a good idea to add it to the repo alongside the csv.

Charles5150, 2026-09-12:
Since this still comes up in searches, and the spreadsheet @harvie256 mentions is indeed gone for good now:

I maintain a fork where the README is the guide that was being asked for here. It covers installing the tools, getting into DFU and flashing, and a reference table for every field in the configuration, which is the part the spreadsheet used to carry in its styling: https://github.com/Charles5150/midi-commander-custom

Two things that may matter to you specifically. There is a desktop configurator, so the spreadsheet is not needed at all any more, and a demo configuration is included that uses every feature, so you can open it and see how something is set up rather than guessing at it.

And the freezes you mentioned are fixed. The firmware called an error handler that disabled interrupts and span forever whenever the MIDI transmit buffers filled up, which is why it locked until a power cycle. A button with several commands, pressed quickly a few times, was enough to reach it.