harvie256/midi-commander-custom #28: From Zero to Custom - A Guide of how to use this software for noobies

ffabreti, 2023-04-11:
### So, you have gotten a MIDI Commander...
...and you have hit lots of limitations, so frustrating to get a good piece of hardware with a crappy software. But wait, that's not all, MeloAudio site seens to be heading the same crappy way.
Enters midi-commander-custom. Harvie256 was kind enough to share his knowledge of reverse engineering poor firmwares and improving over it. But where do you begin ? You know nothing about C++ or Python and you just want to make music!  Keep reading...

### Firmwhat ?
First of, you need to understand a firmware is a bridge between a hardware and another software. In this case, when you press a button with you feet, the firmware will send a MIDI command thru a physical interface (5 pin MIDI or USB) to a computer, an Ipad, an amplifier or any other piece of software inside a hardware you want to control.

The problem with MIDI Commander is that it's firmware has very little functionality and installing midi-commander-custom will allow you to overcome those limitations.

According to Harvie256, this software is written in an area not used by the original firmware, what makes very easy to revert the changes if you need to get back to the original initial state.

### How it works...
Great, let's understand how this software works. Midi-commander-custom is made of 2 parts: 
- a software that controls the hardware, written in C++ and responsible for interacting with the electronics inside like buttons, output ports, microcontrollers, etc.
- another software that configures the button actions, written in Python and responsible by customizing the ways the hardware will behave when a user interacts with the hardware.

The most important piece to the end-user is the Python part. We will focus on that.

continues...

andrema2, 2023-04-12:
This is going to be great.... Looking forward for the rest

angeloregis49, 2023-07-27:
Good job guys! I am also looking forward for the rest of the explanation.

eliericha, 2023-08-18:
Hello folks,

In my fork of the project I expanded the documentation a bit on the topics of setting up Python and loading the CSV config and the firmware. It's [over here](https://github.com/eliericha/midi-commander-custom/tree/main). Hope this helps!

I also created a pull request to get it into the main project but I suspect that harvie256 might be busy these days as he hasn't been very active here recently :)

harvie256, 2023-08-18:
Thanks @eliericha I've merged the pull request.

Yes sorry, I've been in a field camp for the last couple of months so haven't been active of any projects!

eliericha, 2023-08-19:
No worries, thanks @harvie256 !