harvie256/midi-commander-custom #31: Midi in for led's

hellemmm, 2023-08-24:
Hi! As i use midi controllers - i am very glad to meet any thet have an option to receive midi and control it's LED's. Does your script support that feature? Can controller receive midi? Are its LEF controllable?
Thanks)

harvie256, 2023-08-24:
No there's currently no function to control the LEDs via midi input.

Can it receive Midi? Yes it can, but current doesn't process anything.

I'm not sure how this would work with all the other functionality. Could you give a user story of how you would use it? And what you'd expect the buttons to then do?

closed 2023-08-29 by harvie256

EskilSeter, 2025-04-15:
I would love this too.

I use the Midi Commander to control Gig Performer. Setting the switches to toggle internally is well and good, but when I switch variations (basically presets) in Gig Performer, I would need to be able to update the LED and toggle state of each switch to correspond to the current state of the selected variation/preset.

Right now I've just set all switches to momentary, which works well, but I have no visual indication of the actual state of a given effect without looking at my laptop screen.

Charles5150, 2026-09-26:
For anyone arriving here from a search (the Gig Performer case above especially): this exists now in the fork I maintain, built on arasan95's, which builds on this one.

With `LED_Feedback` on, a CC or a note arriving over USB lights or darkens the toggle buttons that send it. The LED, the display and what the next press sends all follow, and nothing is sent back, so a host that echoes messages causes no loop. If the host reports the state on another CC or with other values, a `Listen` command on the button follows that instead.

Docs: https://github.com/Charles5150/midi-commander-custom/blob/main/docs/manual/en/06-commands.md#listening-on-another-cc and the `LED_Feedback` row in https://github.com/Charles5150/midi-commander-custom/blob/main/docs/manual/en/12-configuration-file.md
Ready built image: https://github.com/Charles5150/midi-commander-custom/releases/latest