harvie256/midi-commander-custom #32: Expression Pedals?

pandaskater, 2023-11-04:
Any update on when expression pedal may be implemented? It's just the one step away from being perfect...
Or is already there and I just can't figure it?
Cheers..

Charles5150, 2025-02-25:
+ 1000000

MrHaroldA, 2026-02-06:
This fork should have expression pedal support: https://github.com/ivaylomil/midi-commander-custom

And there's a fork of that fork that add a GUI: https://github.com/arasan95/midi-commander-custom

Charles5150, 2026-09-12:
Following up on this one, since it was left at "some fork should have it": expression pedals are fully supported now, and calibrated from the configurator rather than by guessing numbers.

Per pedal you get the CC number, the MIDI channel, calibrated heel and toe end points, a response curve and an invert switch. The editor shows the pedal position and the CC it is sending in real time, and calibrating is a matter of pressing a button, sweeping the pedal heel to toe a couple of times, and pressing done. The end points are filled in with a small margin so 0 and 127 are always reached.

Each pedal can also act as two extra footswitches: reaching the toe, or coming back to the heel, triggers a button of the current bank while the pedal still sends its CC as usual. https://github.com/Charles5150/midi-commander-custom