harvie256/midi-commander-custom #33: Go back to the meloaudio software

H4ndrax, 2023-11-06:
Please, I need to know if there´s a way to factory reset and go back to the software that was originally in the midi commander, i´ll be very grateful

rahmatismail, 2023-11-19:
I have all meloaudio firmware backup. But I could not ensure if these firmwares will satisfy your need

H4ndrax, 2023-11-19:
> I have all meloaudio firmware backup. But I could not ensure if these firmwares will satisfy your need
Could you please share those files and explain how to do it?

MrHaroldA, 2026-02-06:
You can find the latest Melo or Harley Benton firmware here: https://web.archive.org/web/20220315070346/https://meloaudio.com/pages/manual

You can flash it with dfu-util: `dfu-util -D ~/Downloads/TSMIDI_DFU_V3_STD2.dfu`